hku-mars/loam_livox #127: Does it available for HAP?

Minsu-Kim-KNU, 2023-04-03:
Hi all,

I try to use this package with LIVOX HAP LiDAR.

I set up my LiDAR as follows: 

frame_id = base_link
topic type: sensor_msgs/Pointcloud2

I already tried to use this code with livox.launch and _largescale.launch with modified by myself, but it does not work.

points are received well but aft_mapped_to_init topic which one is the odometry data was not caculated? published well.
it could not update its location.

Does anybody have the same problem with this or had such an experience?

WH30, 2023-08-02:
> Same question, has this problem been solved?

FuturoAbetura, 2023-10-10:
Yes, I had the same problem with Livox HAP TX. When using livox.launch for real-time scanning and mapping, the previously scanned point cloud disappeared. But it will not disappear when using rosbag.launch to record and scan (although the mapping effect is very poor). In addition, try to move at a low speed during lidar movement.