hku-mars/loam_livox #131: CustomMsg and sensor_msgs/PointCloud2

ORBslam2-3, 2023-08-15:
Both the topic sent by mid360 radar and the topic received by the code are: /lidar/livox, but mid360 uses a new driver and relies on SDK2, and the data type is CustomMsg, but this code is different from its data type, how should it be done? 
When I start the code, playing rosbag will appear, as follows：
[ERROR] [1692094106.143429146]: Client [/livox_scanRegistration] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver2/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.

Tianci-Wen, 2024-01-03:
I have same error while using MID-360 with loam_livox

anc95, 2025-03-11:
@Tianci-Wen have you fixed it, or do you know other framework working on mid360 with loop closure checking